dirkgroenen/hass-evse-load-balancer #25: Incompatible with Zaptec integration ≥ 0.7.4 (entity keys renamed)

Ventilgummi, 2025-06-16:
**Describe the bug**  
Since upgrading the *Zaptec* custom integration to **v0.7.4**, EVSE Load Balancer no longer limits or reads current on my Zaptec Go charger. The Zaptec refactor split the previously aggregated attributes into individual entities, so the entity translation keys hard-coded in EVSE are no longer present.

**Affected versions**

- Zaptec integration: **v0.7.4 and newer**  
- hass-evse-load-balancer: **current master (dca9b73)** ← adjust if you use another commit/tag  

**Entities before v0.7.x** (what EVSE still expects)

- `charger_max_current`  
- `available_current`  

**Entities after v0.7.x**

- `sensor.<name>_charge_current_set`  
- `sensor.<name>_available_current_phase1/2/3`  

| Key in EVSE code      | New equivalent entity                          |
|-----------------------|------------------------------------------------|
| `charger_max_current` | `sensor.*_charge_current_set`                  |
| `available_current`   | `sensor.*_available_current_phase1` (and _p2/3)|

`operating_mode` is unchanged and still works.

**Steps to reproduce**

1. Install Zaptec ≥ 0.7.4 and enable its new entities (`*_charge_current_set`, etc.).  
2. Install & configure EVSE Load Balancer for the charger.  
3. Start charging — the current limit never updates; Home Assistant logs show `None` or “entity not found”.  

**Expected behavior**  
EVSE should detect the new Zaptec entities and adjust the charger’s current limit accordingly.

**Proposed fix**

_Update the translation map in `custom_components/evse_load_balancer/charger/zaptec_charger.py`:_

```python
class ZaptecEntityMap:
    # Was: ChargingCurrent = "charger_max_current"
    ChargingCurrent = "charge_current_set"

    # Was: MaxChargingCurrent = "available_current"
    MaxChargingCurrent = "available_current_phase1"  # read one phase; extend as needed

Ventilgummi, 2025-06-17:
Hi,
Thanks for replying. I'm trying to adjust it myself, however I've encountered issues with the meter as well (I'm using a p1ib that exposes kW, not W as current integration expects). Im working on adding the p1ib meter in parallel with adjusting the Zaptec charger.
ill happily test, but you could also wait and let me try to adjust and post potential fix.

dirkgroenen, 2025-06-18:
More than happy to wait and review your implementation - thanks @Ventilgummi 🙌 
If you need some inspiration on adding chargers I just opened another PR #26 which you might be able to use as reference. 

If you need any help just let me know!

Ventilgummi, 2025-06-30:
Hi @dirkgroenen 
Sorry for the late reply. You are completely correct in your last post. I was mistaken. 
I'll happily help out with testing!

Currently I'm trying to add the P1IB meter - which is proving difficult. Hopefully I can fix this within the coming week.

runemoennike, 2025-07-01:
If you need me to test anything, I also have a Zaptec Go charger.

The integration seems to work for me, but I haven't been able to test itnit actually adjusts the currents. I have a "AMS leser" meter connected via Mqqt. @Ventilgummi I endes up using a "helper" entity to convert the loads from watts to kW, otherwise I got insane values from this integration. It seems odd that you have to do the opposite conversion?

dirkgroenen, 2025-07-01:
> I’ve been unable to test my own Zaptec charger changes because I can’t get the meter to work. When I use a custom meter, my P1IB reports **kW**, but **W** is expected. To try to remedy this I created templates like:

Like @runemoennike also mentioned the integration actually expects **kW**, so you shouldn't have to do any conversion. 
The only workaround you need is to provide a bogus (always zero) sensor for "production", as your meter only seems to return an "active power" (being either positive or negative). 

> Here’s my **Zaptec charger adapter**, reworked to support the new entity model and fallback directly to HA state if needed:

Not sure why any rework would be needed if we also conclude that the entities exposed are indeed correct. I wouldn't touch the Zaptec integration at this stage unless you can confirm it's not working for you. In that case it might actually be something else though, as @runemoennike has his working (from my understanding).

> Then I spent (countless) hours with ChatGPT trying to build a full **P1IB integration** instead — and ended up with this version that _almost_ works:
> 
> **P1IB MQTT Meter – per-phase power support (rev-4)**

Now for the meter: that one indeed is going to be a bit more challenging as you need to implement MQTT communication according to the integration. There's already an MQTT base class used for charging, which we might be able to re-use (and further abstract into a MQTT comms class) so it can be used for both chargers and meters. 

That being said: as we only need meter values here I would actually _not_ try to implement native MQTT communication in the meter, but instead try to understand how the entities are being exposed into home assistant. If we know the logic behind that we can simply use the entities state, instead of having to use MQTT's value - which is also more per design. 

Let me have a look at it as well to see if I can help setting up some base which you can then further test and finetune. ChatGPT if fun, but for these architectural implementations it barely produces reliable results from my experience 😅

Ventilgummi, 2025-07-01:
Can confirm that I am plugged in and charging.
Zaptec integration (Version 0.7.4), my "operating_mode" seems to be called "sensor.laddbox_laddstatus_charger_mode" with available states: Unknown, Disconnected, Waiting, Charging, Charge done. It is currently "Charging" (however, as I have home assistant in swedish it says "Laddar").

dirkgroenen, 2025-07-01:
@Ventilgummi just to give me a bit more, can you check the state and other attributes in HASS' `Developer Tools -> States`? 

_Example_
[image]

Ventilgummi, 2025-07-01:
Absolutely, I assume you mean from my charger?
Hope I haven't redacted anything necessary

Entity: [binary_sensor.laddbox_charger]
Attributes:
id: 15f1450c-fb6a-4d8c-8505-217cbc93bb30
device_id: ZAG060824
mid: ZAG060824
name: Laddbox
serial_no: Laddbox
active: true
device_type: Apollo
operating_mode: Connected_Charging
is_online: true
created_on_date: 2025-05-22T10:49:02.967
circuit_id: 889a0973-37d4-4248-bfda-75b21ee77074
current_user_roles: Owner, User, None
pin: 1391
property_pin_offline_phase: false
property_authentication_disabled: false
has_sessions: true
property_offline_phase_override: 0
signed_meter_value_kwh: 279.289
signed_meter_value: 
FV: '1.0'
GI: ZAPTEC GO
GS: ZAG060824
GV: 2.4.2.4
PG: F1
RD:
  - TM: 2025-07-01T10:00:00,000+00:00 R
    RV: 279.289
    RI: 1-0:1.8.0
    RU: kWh
    RT: AC
    ST: G

installation_name: Boaryd 4
installation_id: redacted
authentication_type: Native
is_authorization_required: false
offline_mode: 0
capabilities: {"CommunicationModes":["LTE","WiFi"],"DeviceType":"Go","MeterCalibrated":false,"OcppVersions":["V1_6"],"SchemaVersion":"2.1"}
authentication_required: false
transmit_interval: 3600
pulse_interval: 180
communication_mode: Wifi
permanent_cable_lock: true
hmi_brightness: 0.700
temperature_internal5: 63.066
temperature_e_meter_phase1: 47.606
temperature_e_meter_phase2: 48.280
temperature_e_meter_phase3: 48.492
temperature_tm: 43.779
temperature_tm2: 45.068
temperature_session_max_air: 35.094
temperature_session_max_board: 28.461
humidity: 14.710
voltage_phase1: 242.573
voltage_phase2: 243.999
voltage_phase3: 247.938
current_phase1: 8.735
current_phase2: 8.167
current_phase3: 0.049
charger_max_current: 16
charger_min_current: 6
total_charge_power: 3507.475
set_phases: 4
max_phases: 3
charger_offline_phase: 4
charger_offline_current: 16.000
max_current_configuration_source: 2
switch_position: 0
charge_current_installation_max_limit: 16
stand_alone_current: 16.000
phase_rotation: 4
total_charge_power_session: 3.329
charge_mode: 6
charge_current_set: 8.996
charger_operation_mode: Connected_Charging
is_enabled: 1
is_stand_alone: 0
cable_type: 20
network_type: TN_3_Phase
final_stop_active: 0
session_identifier: redacted
completed_session: 
SessionId: redacted
Energy: 0.247
StartDateTime: '2025-07-01T09:27:54.039349Z'
EndDateTime: '2025-07-01T09:46:42.341038Z'
ReliableClock: true
StoppedByRFID: false
AuthenticationCode: ''
FirstAuthenticatedDateTime: ''
SignedSession:
redacted
override_network_type: 4
nr_of_charge_cards: 1
authentication_list_version: 1
location: SWE
time_zone: Europe/Stockholm
routing_id: default
charge_point_name: Laddbox
warnings: 0
diagnostics_mode: 0
internal_diagnostics_log: #10 Cloud restart command
diagnostics_string: 39d 00h24m10s T_EM: 47.61 48.28 48.49  T_M: 43.78 45.07   V: 0.35 0.20 0.77   I: 0.03 0.02 0.03  0.085kW 0.000kWh CP: 12.0 C6 CM3 MCnt:2714060 Rs:9 Rc:0 NTPs: 79
communication_signal_strength: -42.000
mcu_reset_source: 1
esp_reset_source: 3
certificate_version: 8
security_log: [2025-05-22T11:24:39+0000] OTA             : Accepted
session_controller: 0
ocpp_native_url: 
ocpp_native_cb_id: 
ocpp_native_authorization_key_from_zaptec: 0
ocpp_native_security_profile: 0
ocpp_native_connected: 0
smart_mainboard_software_application_version: 2.4.2.4
smart_mainboard_software_bootloader_version: 6
smart_computer_software_application_version: 2.4.2.4
source_version: 2.4.2.4_Go
hw_id_main_board: 5
hw_id_power_board: 4
mac_wi_fi: 30:c9:22:e4:20:74
current_in_maximum: 16
default_offline_phase: 4
default_offline_current: 16
current_firmware_version: 2.4.2.4
available_firmware_version: 2.4.2.4
firmware_update_to_date: true
device_class: connectivity
icon: mdi:cloud
friendly_name: Laddbox Charger

runemoennike, 2025-07-01:
I don't know if this is useful at this point, but I had overlooked the EVSE integration's state field. I just plugged in my car to the charger, and while the Zaptec integration reports "charging" on the "charger_mode" entity, the EVSE state remains "waiting for charger". The available current values look right, however.

Is there another branch I should test, or more information I can provide?

runemoennike, 2025-07-01:
After changing to that branch, the Load Balancing State indeed changes to "Monitoring Loads" when I start charging. However, it doesn't seem to go back to "Awaiting charger" when I stop the charging?

Here is the debug log filtered to the EVSE lines: [home-assistant_evse_load_balancer_2025-07-01T20-22-32.315Z.log](https://github.com/user-attachments/files/21007101/home-assistant_evse_load_balancer_2025-07-01T20-22-32.315Z.log)

Some other observations:

1. The log has some lines indicating a type conversion error: `Could not convert current limit '16.0' to number` -- that 16 is probably from the current limit on the charger? My phase limit for the house is 25, but the charger is limited to 16 per phase.
2. The system log has the message `Module custom_components.evse_load_balancer.meters.custom_meter is logging too frequently. 200 messages since last count`, but maybe that's related to the extra debug messages.
3. I noticed the "Logbook" panel on the EVSE Load Balancer device page constantly shows a new "EVSE Load Balancer Last check changed to 1" message coming in:

[image]

But maybe that's related to the same thing :)

Also, after changing branch I'm unable to reach my Home Assistant via the Nabu Casa cloud service, but I cannot see how that can possibly be related.

dirkgroenen, 2025-07-01:
Thanks @runemoennike - I'll have a look at your message tomorrow and see if  it explains a few things.

I did add quite a verbose message indeed, so that could explain one of those messages. Feel free to disable the debug logging or revert back to the stable release in the meantime ofc.

dirkgroenen, 2025-07-02:
@runemoennike had a look at the logs and made two changes to the branch (commit `7396affce92c73064bc82851aa2abba78804be8c`):
1. That `16.0` was coming from the max limit indeed, but seemed to be returned as either a float or string-float. I've fixed the conversion to have it be converted to int correctly. 
2. I've removed the debug logs again, as I noticed I included the "charger finished" state in `can_charge`, which is indeed a bit weird as there's nothing to monitor or assign when a charger is idling at a finished state. 

As for the "Last Check" logbook entry: I'm aware of that indeed and actually haver so far not been able to tell HASS to exclude that one. Tried all the things I was able to find in the dev docs, but no success.  I'm not sure if you actually use the sensor, as I personally ended up deleting it and am actually tempted to remove it from the codebase as well. 

If you want to run another test feel free to run:
```
action: update.install
data:
  version: 7396affce92c73064bc82851aa2abba78804be8c
target:
  entity_id: update.evse_load_balancer_update
```

If all is good I'll merge and release.

runemoennike, 2025-07-02:
I tested and it works great now 🙂 

The Last Check entity was also disabled by default, as you said, after I recreated the device.

I was unable to select my dummy helper entity in the drop-down for production entities in the phases, and had to select an entity from my meter instead (which strangely only has a single production entity for all phases). The first version I tried, 1.0.0 I think, let me select it just fine. Idk if it's another quirk, and in any case I don't have production so it works fine for me now.

Thanks!

Ventilgummi, 2025-07-02:
Sounds awesome, will try it as soon as I can as well! Thanks for this 🙏