dirkgroenen/hass-evse-load-balancer #27: [meter] Add support for HomeWizard P1

dirkgroenen, 2025-06-18:
Add support for HomeWizard's P1 meter:
- https://www.home-assistant.io/integrations/homewizard/
- https://github.com/home-assistant/core/blob/dev/homeassistant/components/homewizard/sensor.py

jochimsergoyne, 2025-06-24:
The homewizard meter is visible. I selected it and have 1 device and 5 entities.

[image]

I do have an error though in my logbook, but that could be because my homewizard integration was not fully loaded (unsure?)

[image]

dirkgroenen, 2025-06-24:
Thanks @jochimsergoyne! Would you mind showing me the available entities produced by the HomeWizard integration? For example via the "Settings -> Devices -> Entities" tab, and then filtering on the "HomeWizard" integration:

[image]

I'm particularly interested in the Active Power sensors, cause it seems like one of the sensors that's being expected (and should be available according to the HomeWizard integration's code) is not available.

dirkgroenen, 2025-06-24:
Thanks. I think I've got the problem, seems like my attempt on retrieving the sensors by their translation key does not work for Home Assistant. I'll have to make a few changes to the code, but before I do can you confirm my assumptions by making a screenshot of the "Stroom Fase 1" sensor details? Specifically the "Entity ID" I'm interested in. 

[image]

jochimsergoyne, 2025-06-24:
sensor ID: sensor.p1_meter_stroom_fase_1
[image]

dirkgroenen, 2025-06-24:
Alright, thanks. I'm going to make some changes in a bit and let you know once they are pushed. If you can then re-run the update steps and test again it should hopefully work. 

I'll let you know when.

dirkgroenen, 2025-06-24:
@jochimsergoyne would you mind trying to update using the below (latest commit on the branch): 

```yaml
action: update.install
data:
  version: 306b1ba2d92a67ea7377d8c2988629d446a9f6e1
target:
  entity_id: update.evse_load_balancer_update
```

And see what it does?

jochimsergoyne, 2025-06-24:
removed integration, updated according to your message, restarted HA, re-added integration. Got the following message:

Deze fout is ontstaan door een aangepaste integratie.

Logger: homeassistant
Bron: custom_components/evse_load_balancer/coordinator.py:301
integratie: EVSE Load Balancer (documentatie, problemen)
Eerst voorgekomen: 16:35:23 (1 gebeurtenis)
Laatst gelogd: 16:35:23

Error doing job: Exception in callback _TrackTimeInterval._interval_listener() (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1664, in _interval_listener
    self.hass.async_run_hass_job(self._run_job, dt_util.utcnow(), background=True)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 938, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/config/custom_components/evse_load_balancer/coordinator.py", line 222, in _execute_update_cycle
    self._update_charger_settings(allocation_result)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evse_load_balancer/coordinator.py", line 293, in _update_charger_settings
    self._emit_charger_event(EVENT_ACTION_NEW_CHARGER_LIMITS, new_limits)
    ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evse_load_balancer/coordinator.py", line 301, in _emit_charger_event
    ATTR_DEVICE_ID: self._device.id,
                    ^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'id'

dirkgroenen, 2025-06-24:
That seems to be related to #30, which I discovered this morning as well. It shouldn't prevent the integration from being though. Can you confirm the integration adds a device and starts monitoring?

dirkgroenen, 2025-06-27:
Hi @jochimsergoyne, have you managed to get this working by any change? Planning on rolling out another version in the next few days and would love to include the HomeWizard integration. 🙂

closed 2025-06-27 by dirkgroenen