dirkgroenen/hass-evse-load-balancer #42: Not working with custom meter (P1IB)

Ventilgummi, 2025-07-16:
### Describe the bug
When I enable **custom_meter** with my P1 Interface Bridge (P1IB), every balancing cycle fails and the log loops through:

    Failed to parse state unavailable for entity sensor.p1ib_active_power_plus_l1
    Missing states for one of phase Phase.L1 while balancing
    Available current for phase 'l1' is None. Cannot proceed with balancing cycle.

The entities the balancer looks for (e.g. `sensor.p1ib_active_power_plus_l1`) all exist in Home Assistant and show numeric values in **Developer Tools → States**, yet the integration keeps reading them as `unavailable`.
What's interesting is that the diagnostic part of the load balancer seems to be reading my available current correctly, see attached screenshot (Swedish, but still)

[image]

### Charger
Zaptec integration - which you helped fix in a previous bug, thanks! 👍 

### Troubleshooting already tried
* Confirmed each sensor (e.g. sensor.p1ib_active_power_plus_l1) shows a numeric value (e.g. 0.238 kW) in **Developer Tools → States**.

Happy to help with more debug logs/etc as needed! 

Thanks for your work on this!

dirkgroenen, 2025-07-19:
That's interesting indeed 🤔  Would you mind passing me over your full debug logs? Ideally from the moment the integration is enabled, and with a few cycles of those error logs you're referring to.

Ventilgummi, 2025-08-06:
Hi,

Sorry for the terribly slow response (I've been on vacation, and the weather has been nice!)

Here's the debug logs:
2025-07-16 02:02:51.814 INFO (MainThread) [custom_components.evse_load_balancer] Setting up entry with meter 'CustomMeter' and charger 'ZaptecCharger'
2025-07-16 02:02:51.814 INFO (MainThread) [custom_components.evse_load_balancer.power_allocator] Added charger 01JZWEG67MPVAXGR5H5PQH9T4E to PowerAllocator
2025-07-16 02:02:51.814 DEBUG (MainThread) [custom_components.evse_load_balancer] EVSE Load Balancer initialized for redacted
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_active_power_plus_l1
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_active_power_minus_l1
2025-07-16 02:02:51.841 WARNING (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] Missing states for one of phase Phase.L1: consumption: None, production: None. Are the entities enabled?
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_voltage_l1
2025-07-16 02:02:51.841 WARNING (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] ('Missing states for one of phase Phase.L1: active_power: None,', ' voltage_state: None. Are the entities enabled?')
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_active_power_plus_l2
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_active_power_minus_l2
2025-07-16 02:02:51.841 WARNING (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] Missing states for one of phase Phase.L2: consumption: None, production: None. Are the entities enabled?
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_voltage_l2
2025-07-16 02:02:51.841 WARNING (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] ('Missing states for one of phase Phase.L2: active_power: None,', ' voltage_state: None. Are the entities enabled?')
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_active_power_plus_l3
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_active_power_minus_l3
2025-07-16 02:02:51.841 WARNING (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] Missing states for one of phase Phase.L3: consumption: None, production: None. Are the entities enabled?
2025-07-16 02:02:51.841 DEBUG (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] State not found for entity sensor.p1ib_voltage_l3
2025-07-16 02:02:51.841 WARNING (MainThread) [custom_components.evse_load_balancer.meters.custom_meter] ('Missing states for one of phase Phase.L3: active_power: None,', ' voltage_state: None. Are the entities enabled?')

And this is an example from the error log:

Loggare: custom_components.evse_load_balancer.meters.custom_meter
Källa: custom_components/evse_load_balancer/meters/custom_meter.py:87
integration: EVSE Lastbalanserare (dokumentation, ärenden)
Inträffade först: 02:02:59 (36 förekomster)
Senast loggade: 02:03:01
Failed to parse state unavailable for entity sensor.p1ib_active_power_minus_l2: could not convert string to float: 'unavailable'
Failed to parse state unavailable for entity sensor.p1ib_voltage_l2: could not convert string to float: 'unavailable'
Failed to parse state unavailable for entity sensor.p1ib_active_power_plus_l3: could not convert string to float: 'unavailable'
Failed to parse state unavailable for entity sensor.p1ib_active_power_minus_l3: could not convert string to float: 'unavailable'
Failed to parse state unavailable for entity sensor.p1ib_voltage_l3: could not convert string to float: 'unavailable'
Traceback (most recent call last):
  File "/config/custom_components/evse_load_balancer/meters/custom_meter.py", line 87, in _get_state
    return float(state_value)
ValueError: could not convert string to float: 'unavailable'