dirkgroenen/hass-evse-load-balancer #46: add support for Shelly EM3

kljakobsen, 2025-07-21:
Add support for the Shelly local integration, as a smart meter.

https://www.home-assistant.io/integrations/shelly

The integration created a device - <device name>, that has 3 related devices, 1 for each phase. these devices current (a) can be read in near real time

dirkgroenen, 2025-07-26:
Hi @kljakobsen! To help me understand how the Shelly integration exposes such an energy meter, can you share me some screenshots of the device and its exposed states? Ideally also including the `Entities` page filtered on the EM3 device(s)? 

Right now the integration expects a single device to expose 3 phases, but if I understand you correctly the Shelly integration exposes multiple devices all having 1 entity?

kljakobsen, 2025-07-26:
Hi Dirkgroenen,
Sure.. 
Devices:
[image]

Device relation:
[image]

Entities:
[image]

jonasman, 2025-09-20:
hi i got a shelly 3EM, im not sure what is the issue here, but the main device exposes the total combiner power in one property too

dirkgroenen, 2025-10-18:
For everyone following this thread interested in an EM3 integration: I'm a bit busy these months with work so I don't have enough time to look into this right now. Feel free to open up a PR of course, which I'm happy to review. 

> But the available power values looks a bit weird?
> Could it be some value troubles since my 3EM sensors are in W and not kW?

Correct @VVVVVVVip, sensor values are [expected to be in kW](https://github.com/dirkgroenen/hass-evse-load-balancer?tab=readme-ov-file#advanced-configuration). You can always add a Helper Sensor to convert it from W to kW.