dirkgroenen/hass-evse-load-balancer #60: Charger current not increased when available current increases

marcuswahlstam, 2025-08-31:
Setup: Zaptec Go charger and Home Wizard P1 meter

Since 1.0.4 the load balancing works perfectly with my setup when the available current drops below 0. But when the available current increases, the charging current is never increased after 15 minutes. The integration does not even log that it tries to increase the current.

Is this a general bug or just a bug with my setup?

Oh, and the version number states 1.0.3 on the integration page in HA, even though it's 1.0.4. I've checked the files, and files are the fixed ones for 1.0.4.

dirkgroenen, 2025-09-01:
Would you mind running the integration in debug mode and pasting me a full output of its logs. Preferably from the moment before you plug-in the car until the point where it decreases and (should) increase. 

> Oh, and the version number states 1.0.3 on the integration page in HA, even though it's 1.0.4. I've checked the files, and files are the fixed ones for 1.0.4.

Good one, forgot to change those indeed. Too late now unfortunately 🙈

marcuswahlstam, 2025-09-03:
Here is the log
```
2025-09-03 01:07:20.979 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] New charger settings: {<Phase.L1: 'l1'>: 8, <Phase.L2: 'l2'>: 8, <Phase.L3: 'l3'>: 8}
2025-09-03 01:07:20.979 INFO (MainThread) [custom_components.evse_load_balancer.coordinator] Emitted charger event: action=new_charger_limits, new_limits={<Phase.L1: 'l1'>: 8, <Phase.L2: 'l2'>: 8, <Phase.L3: 'l3'>: 8}
2025-09-03 01:07:20.981 DEBUG (MainThread) [custom_components.evse_load_balancer.power_allocator] Updated applied current for charger 01K3ZDF196KT8DF7VZDGMV10SA: {<Phase.L1: 'l1'>: 8, <Phase.L2: 'l2'>: 8, <Phase.L3: 'l3'>: 8}
2025-09-03 01:07:30.989 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] Charger settings was updated too recently. Last update: 1756854440, current time: 1756854450. Configured delay: 15.0 minutes
2025-09-03 01:07:51.012 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] New charger settings are lower, apply ignoring user setting. Last settings: {<Phase.L1: 'l1'>: 8, <Phase.L2: 'l2'>: 8, <Phase.L3: 'l3'>: 8}, new settings: {<Phase.L1: 'l1'>: 5, <Phase.L2: 'l2'>: 5, <Phase.L3: 'l3'>: 5}
2025-09-03 01:07:51.012 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] New charger settings: {<Phase.L1: 'l1'>: 5, <Phase.L2: 'l2'>: 5, <Phase.L3: 'l3'>: 5}
2025-09-03 01:07:51.012 INFO (MainThread) [custom_components.evse_load_balancer.coordinator] Emitted charger event: action=new_charger_limits, new_limits={<Phase.L1: 'l1'>: 5, <Phase.L2: 'l2'>: 5, <Phase.L3: 'l3'>: 5}
2025-09-03 01:07:51.014 DEBUG (MainThread) [custom_components.evse_load_balancer.power_allocator] Updated applied current for charger 01K3ZDF196KT8DF7VZDGMV10SA: {<Phase.L1: 'l1'>: 5, <Phase.L2: 'l2'>: 5, <Phase.L3: 'l3'>: 5}
2025-09-03 03:24:52.421 INFO (MainThread) [custom_components.evse_load_balancer.power_allocator] Manual override detected for charger. New requested current: {<Phase.L1: 'l1'>: 6, <Phase.L2: 'l2'>: 6, <Phase.L3: 'l3'>: 6}
2025-09-03 03:25:22.449 INFO (MainThread) [custom_components.evse_load_balancer.power_allocator] Manual override detected for charger. New requested current: {<Phase.L1: 'l1'>: 13, <Phase.L2: 'l2'>: 13, <Phase.L3: 'l3'>: 13}
2025-09-03 03:26:07.457 INFO (MainThread) [custom_components.evse_load_balancer] Setting up entry with meter 'HomeWizardMeter' and charger 'ZaptecCharger'
2025-09-03 03:26:07.457 INFO (MainThread) [custom_components.evse_load_balancer.power_allocator] Added charger 01K3ZDF196KT8DF7VZDGMV10SA to PowerAllocator
2025-09-03 03:26:07.457 DEBUG (MainThread) [custom_components.evse_load_balancer] EVSE Load Balancer initialized for 01K3ZDF196KT8DF7VZDGMV10SA
2025-09-03 03:26:08.458 INFO (MainThread) [custom_components.evse_load_balancer.power_allocator] Charger initialized with limits: {<Phase.L1: 'l1'>: 13, <Phase.L2: 'l2'>: 13, <Phase.L3: 'l3'>: 13}
2025-09-03 05:27:26.762 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] New charger settings: {<Phase.L1: 'l1'>: 10, <Phase.L2: 'l2'>: 10, <Phase.L3: 'l3'>: 10}
2025-09-03 05:27:26.762 INFO (MainThread) [custom_components.evse_load_balancer.coordinator] Emitted charger event: action=new_charger_limits, new_limits={<Phase.L1: 'l1'>: 10, <Phase.L2: 'l2'>: 10, <Phase.L3: 'l3'>: 10}
2025-09-03 05:27:26.764 DEBUG (MainThread) [custom_components.evse_load_balancer.power_allocator] Updated applied current for charger 01K3ZDF196KT8DF7VZDGMV10SA: {<Phase.L1: 'l1'>: 10, <Phase.L2: 'l2'>: 10, <Phase.L3: 'l3'>: 10}
2025-09-03 05:27:32.767 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] Charger settings was updated too recently. Last update: 1756870046, current time: 1756870052. Configured delay: 15.0 minutes
025-09-03 05:30:50.974 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] New charger settings are lower, apply ignoring user setting. Last settings: {<Phase.L1: 'l1'>: 10, <Phase.L2: 'l2'>: 10, <Phase.L3: 'l3'>: 10}, new settings: {<Phase.L1: 'l1'>: 9, <Phase.L2: 'l2'>: 9, <Phase.L3: 'l3'>: 9}
2025-09-03 05:30:50.974 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] New charger settings: {<Phase.L1: 'l1'>: 9, <Phase.L2: 'l2'>: 9, <Phase.L3: 'l3'>: 9}
2025-09-03 05:30:50.975 INFO (MainThread) [custom_components.evse_load_balancer.coordinator] Emitted charger event: action=new_charger_limits, new_limits={<Phase.L1: 'l1'>: 9, <Phase.L2: 'l2'>: 9, <Phase.L3: 'l3'>: 9}
2025-09-03 05:30:50.975 DEBUG (MainThread) [custom_components.evse_load_balancer.power_allocator] Updated applied current for charger 01K3ZDF196KT8DF7VZDGMV10SA: {<Phase.L1: 'l1'>: 9, <Phase.L2: 'l2'>: 9, <Phase.L3: 'l3'>: 9}
2025-09-03 05:31:02.987 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] Charger settings was updated too recently. Last update: 1756870250, current time: 1756870262. Configured delay: 15.0 minutes
2025-09-03 05:34:31.186 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] New charger settings are lower, apply ignoring user setting. Last settings: {<Phase.L1: 'l1'>: 9, <Phase.L2: 'l2'>: 9, <Phase.L3: 'l3'>: 9}, new settings: {<Phase.L1: 'l1'>: 8, <Phase.L2: 'l2'>: 8, <Phase.L3: 'l3'>: 8}
2025-09-03 05:34:31.186 DEBUG (MainThread) [custom_components.evse_load_balancer.coordinator] New charger settings: {<Phase.L1: 'l1'>: 8, <Phase.L2: 'l2'>: 8, <Phase.L3: 'l3'>: 8}
[... cut ...]
2025-09-03 06:41:42.322 INFO (MainThread) [custom_components.evse_load_balancer.power_allocator] Manual override detected for charger. New requested current: {<Phase.L1: 'l1'>: 10, <Phase.L2: 'l2'>: 10, <Phase.L3: 'l3'>: 10}
2025-09-03 06:42:05.347 INFO (MainThread) [custom_components.evse_load_balancer.power_allocator] Manual override detected for charger. New requested current: {<Phase.L1: 'l1'>: 13, <Phase.L2: 'l2'>: 13, <Phase.L3: 'l3'>: 13}
```
Is it because EVSE sets the current lower than 6A? My car (Model 3) stops charging if the current is 5A or lower.
At 03:35 and 06:01 I increased the current to 13A manually.

marcuswahlstam, 2025-09-14:
Update installed and I'll try to try it out today!

marcuswahlstam, 2025-09-17:
Great, my experience is the same as your, it does not increase the current at all for me.

marcuswahlstam, 2025-09-17:
How do you mean? I ran:
```
action: update.install
target:
  entity_id: update.evse_load_balancer_update
data:
  version: 96cdb4c8889e06ef07a602dd3c8b5ccce32a45c2
```
And then restarted HA. Am I supposed to do something else too?

dirkgroenen, 2025-09-17:
Another night, another commit 😉   Time to hit the snooze, but before doing so couldn't resist getting a few fixes out there.  Just updated the "update instructions" in the PR to the latest commit. Did some local simulations manually and was able to find some race-conditions likely causing the bug. 

Feel free to give it another try.

dirkgroenen, 2025-09-18:
Just did some testing and I can confirm it's working correctly now (running an Easee Charger - but that should't matter for this bug).  Hopefully you'll have the same results. 🙏 

[image]

PR #63 (2025-09-19): [#60] Resolve non-increasing charger current 
Fixes #60 by reducing a "previous current" memory which seems to be causing race conditions. Removal of the memorisation should simplify things and hopefully remove the bug. Guards preventing double allocation are still in place by time controls, which should be enough. 

Install:
```
action: update.install
target:
  entity_id: update.evse_load_balancer_update
data:
  version: 94aefcad9d93c450ad3dd97f264bb15980997814
```